mabuchilab/Instrumental #138: Absoulte beginner - some basic help with Instrumental

jlozanol, 2021-04-13:
Hi Team,

Sorry if this is not the right forum to put this question. I'm brand new to general programming and I'm trying to understand how the Instrumental Library works. My short term goal is to create a python program to communicate a Thorlabs Spectrometer CCS175 and I just installed this library.

I just tried to run the list of instruments to see if the spectrometer appeared and I just got the following:

**>>> from instrumental import instrument, list_instruments
>>> paramsets = list_instruments()**
_Traceback (most recent call last):
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 677, in list_instruments
    inst_list.extend(list_visa_instruments())
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 621, in list_visa_instruments
    return list(gen_visa_instruments())
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 567, in gen_visa_instruments
    rm = visa.ResourceManager()
AttributeError: module 'visa' has no attribute 'ResourceManager'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\s3853339\Miniconda3\lib\site-packages\instrumental\drivers\__init__.py", line 678, in list_instruments
    except visa.VisaIOError:
AttributeError: module 'visa' has no attribute 'VisaIOError'_

I'm not sure what I'm doing wrong of if I didn't install any other library. I can see visa and pyvisa are already installed in my base env, so I don't know where to go from here.

As I said, I'm brand new and there are a lot of things I really don't understand. Do you know where I could get some information or some examples on how to use the Instrumental library? (my google searches haven't been successfull on finding something useful so far).

Again, so sorry if this is not the right place to ask this question!

Cheers,

jondoesntgit, 2021-04-13:
There are several packages that are named "visa". It's possible that you've installed the wrong one. This is the one you're after: https://pyvisa.readthedocs.io/en/latest/introduction/getting.html

jlozanol, 2021-04-15:
> There are several packages that are named "visa". It's possible that you've installed the wrong one. This is the one you're after: https://pyvisa.readthedocs.io/en/latest/introduction/getting.html

Thx for the answer @jondoesntgit !
I had already installed pyvisa as I thought it could be the problem. I can see it's in my base env (as well as visa):

```
(base) PS C:\> conda list
# packages in environment at C:\Users\s3853339\Miniconda3:
#
# Name                    Version                   Build  Channel
...
pyvisa                    1.11.3                   pypi_0    pypi
...
visa                      1.0.0                    pypi_0    pypi
```

I tried the basic command that the pyvisa doc indicates:

```
>>> import pyvisa
>>> rm = pyvisa.ResourceManager()
>>> rm.list_resources()
()
```

I imagine there's something wrong with pyvisa that cannot see any devices connected to the computer?
BTW, I have also connected a NI myDAQ device that I have successfully made it work via nidaqmx library.
It seems an issue with pyvisa, maybe?

Thanks heaps!

Jorge

jondoesntgit, 2021-04-16:
Have you installed NI Visa on your machine? PyVISA is a wrapper around some lower-level libraries, and relies on those lower-level libraries being installed. It's been a while since Iast installed, but I hazily remember that there may have been a configuration step in connecting PyVISA with NI VISA (or some other alternative if you're on Linux or something).

jlozanol, 2021-04-28:
> Have you installed NI Visa on your machine? PyVISA is a wrapper around some lower-level libraries, and relies on those lower-level libraries being installed. It's been a while since Iast installed, [...]

Hi @jondoesntgit!

Sorry for the long time replying, I've been so busy with other stuff.

This is what I get from pyvisa-info

```
(base) PS C:\> pyvisa-info
Machine Details:
   Platform ID:    Windows-10-10.0.18362-SP0
   Processor:      Intel64 Family 6 Model 142 Stepping 12, GenuineIntel

Python:
   Implementation: CPython
   Executable:     c:\users\s3853339\miniconda3\python.exe
   Version:        3.9.1
   Compiler:       MSC v.1916 64 bit (AMD64)
   Bits:           64bit
   Build:          Dec 11 2020 09:29:25 (#default)
   Unicode:        UCS4

PyVISA Version: 1.11.3

Backends:
   ivi:
      Version: 1.11.3 (bundled with PyVISA)
      #1: C:\WINDOWS\system32\visa32.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
      #2: C:\WINDOWS\system32\visa64.dll:
         found by: auto
         bitness: 64
         Vendor: National Instruments
         Impl. Version: National Instruments
         Spec. Version: National Instruments
```

I imagine this means NI VISA is currently installed and I shouldn't have this issue...

I then tried the following code, but still not successful:

```
>>> import pyvisa as visa
>>> rm = visa.ResourceManager(r'C:\WINDOWS\system32\visa64.dll')
>>> rm.list_resources()
()
```
I think it will be better if I go to the pyvisa forum for more information.

Thank you very much for your time trying to help me with this!

Cheers,

jlozanol, 2021-04-29:
Hi @jondoesntgit,

I just found the following information on #120.

After checking I discovered I might face the same issue as indicated there:

```
>>> from instrumental.log import log_to_screen
>>> log_to_screen()
<RootLogger root (INFO)>
>>> list_instruments()
[    INFO]2021-04-29 10:41:00,486 nicelib.util: Looking for ('TLCCS.h',) in ('{PROGRAMFILES}\\IVI Foundation\\VISA\\Win64\\Include', '{PROGRAMFILES(X86)}\\IVI Foundation\\VISA\\Win64\\Include')
[    INFO]2021-04-29 10:41:00,488 nicelib.build: Found ['C:\\Program Files\\IVI Foundation\\VISA\\Win64\\Include\\TLCCS.h']
[    INFO]2021-04-29 10:41:00,494 nicelib.build: Parsing and cleaning headers...
[    INFO]2021-04-29 10:41:00,554 instrumental.drivers: Error when importing driver module spectrometers.thorlabs_ccs: <<Unsupported constant type 'long int'>>
```

Unfortunately I'm still a newbie in programing and even in github! So just wanted to ask where I can have the patch that Nate Bogdanowicz mentioned before and where I need to install it on my computer as I don't know where to find it in github:

> I pushed a change to the CCS driver that will instruct NiceLib to choose the right set of typedefs in visatype.h. I'd suggest pulling this new code and deleting your Instrumental\instrumental\drivers\spectrometers_tlccslib.py file so it gets regenerated.

Thank you very much for your help and again apologies if this is not the right way to make this request.

Regards,

jondoesntgit, 2021-04-29:
What you're going to need to do is to clone the current commit of github. That's well documented on the web, but it's gonna be something like:

```
git clone https://github.com/mabuchilab/Instrumental/
```

And then you'll want to install it using 

```
cd Instrumental
pip install -e .
```

jlozanol, 2021-04-29:
Hi!

Thanks for the information, I already installed this latest version as advised:

```
(test) PS C:\Instrumental> dir                                                                                          

    Directory: C:\Instrumental

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        4/30/2021   8:08 AM                docs
d-----        4/30/2021   8:08 AM                examples
d-----        4/30/2021   8:08 AM                images
d-----        4/30/2021   8:14 AM                instrumental
d-----        4/30/2021   8:08 AM                Instrumental_lib.egg-info
d-----        4/30/2021   8:08 AM                tests
d-----        4/30/2021   8:08 AM                tools
-a----        4/30/2021   8:08 AM            214 .gitignore
-a----        4/30/2021   8:08 AM            481 .travis.yml
-a----        4/30/2021   8:08 AM           1679 appveyor.yml
-a----        4/30/2021   8:08 AM            352 AUTHORS
-a----        4/30/2021   8:08 AM           7488 CHANGELOG.rst
-a----        4/30/2021   8:14 AM           7291 lextab.py
-a----        4/30/2021   8:08 AM          35831 LICENSE
-a----        4/30/2021   8:08 AM            214 MANIFEST.in
-a----        4/30/2021   8:08 AM           2904 README.rst
-a----        4/30/2021   8:08 AM             51 readthedocs-requirements.txt
-a----        4/30/2021   8:08 AM            150 setup.cfg
-a----        4/30/2021   8:08 AM           4203 setup.py
-a----        4/30/2021   8:14 AM         175998 yacctab.py

(test) PS C:\Instrumental> pip install -e .                                                                             Obtaining file:///C:/Instrumental
Requirement already satisfied: numpy in c:\users\s3853339\miniconda3\envs\test\lib\site-packages (from Instrumental-lib==0.6) (1.20.2)
Requirement already satisfied: scipy in c:\users\s3853339\miniconda3\envs\test\lib\site-packages (from Instrumental-lib==0.6) (1.6.3)
Requirement already satisfied: pint>=0.7 in c:\users\s3853339\miniconda3\envs\test\lib\site-packages (from Instrumental-lib==0.6) (0.17)
Requirement already satisfied: future in c:\users\s3853339\miniconda3\envs\test\lib\site-packages (from Instrumental-lib==0.6) (0.18.2)
Requirement already satisfied: packaging in c:\users\s3853339\miniconda3\envs\test\lib\site-packages (from pint>=0.7->Instrumental-lib==0.6) (20.9)
Requirement already satisfied: pyparsing>=2.0.2 in c:\users\s3853339\miniconda3\envs\test\lib\site-packages (from packaging->pint>=0.7->Instrumental-lib==0.6) (2.4.7)
Installing collected packages: Instrumental-lib
  Attempting uninstall: Instrumental-lib
    Found existing installation: Instrumental-lib 0.6
    Uninstalling Instrumental-lib-0.6:
      Successfully uninstalled Instrumental-lib-0.6
  Running setup.py develop for Instrumental-lib
Successfully installed Instrumental-lib
(test) PS C:\Instrumental>                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                   [... cut ...]
    py_right = to_py_src(node.right)
  File "C:\Users\s3853339\Miniconda3\envs\test\lib\site-packages\nicelib\process.py", line 1625, in to_py_src
    raise ConvertError("Unsupported constant type '{}'".format(node.type))
**nicelib.process.ConvertError: Unsupported constant type 'long int'**
```

The error I'm more concern atm is the `**ModuleNotFoundError: No module named 'instrumental.drivers.spectrometers._tlccslib'**`, I'm not sure if something is still not installed...

Thank you again for your patience!

jondoesntgit, 2021-04-30:
I think this is an option that you have to supply during installation. @rogers132 would probably know how to invoke it, since he wrote:

https://github.com/mabuchilab/Instrumental/blob/8cfc5a1f8c93a00a5b23881b170b3fb191bb2087/instrumental/drivers/spectrometers/_build_tlccs.py

If I had to guess, you need something like

```
pip install instrumental[spectrometers.tlcc]
```

jlozanol, 2021-04-30:
Ok,

I think I know what happened there... Instrumental driver was called `_build_tlccs.py` instead of `_tlccslib`. I changed the name of that file and now I got this error:

```
Python 3.9.4 (default, Apr  9 2021, 11:43:21) [MSC v.1916 64 bit (AMD64)] :: Anaconda, Inc. on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from instrumental.log import log_to_screen
>>> log_to_screen()
<RootLogger root (INFO)>
>>> from instrumental import instrument, list_instruments
>>> from instrumental.drivers.spectrometers.thorlabs_ccs import CCS
C:\Users\s3853339\Miniconda3\envs\test\Lib\site-packages\visa.py:13: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
  warnings.warn(
[    INFO]2021-04-30 13:03:08,416 nicelib.nicelib: Creating class NiceObject...
[    INFO]2021-04-30 13:03:08,416 nicelib.nicelib: Creating class NiceLib...
[    INFO]2021-04-30 13:03:08,417 nicelib.nicelib: Processing attr '__module__' (<class 'str'>)...
[    INFO]2021-04-30 13:03:08,417 nicelib.nicelib: ...as an ordinary class attribute
[    INFO]2021-04-30 13:03:08,417 nicelib.nicelib: Processing attr '__qualname__' (<class 'str'>)...
[    INFO]2021-04-30 13:03:08,417 nicelib.nicelib: ...as an ordinary class attribute
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: Processing attr '__doc__' (<class 'str'>)...
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: ...as an ordinary class attribute
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: Processing attr '_ffi' (<class 'NoneType'>)...
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: ...as a special NiceLib attribute
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: Processing attr '_ffilib' (<class 'NoneType'>)...
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: ...as a special NiceLib attribute
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: Processing attr '_defs' (<class 'dict'>)...
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: ...as a special NiceLib attribute
[    INFO]2021-04-30 13:03:08,418 nicelib.nicelib: Processing attr '_ret_return' (<class 'nicelib.nicelib.RetHandler'>)...
[    INFO]2021-04-30 13:03:08,419 nicelib.nicelib: ...as a RetHandler
[    INFO]2021-04-30 13:03:08,419 nicelib.nicelib: Processing attr '_ret_ignore' (<class 'nicelib.nicelib.RetHandler'>)...
[    INFO]2021-04-30 13:03:08,419 nicelib.nicelib: ...as a RetHandler
[    INFO]2021-04-30 13:03:08,419 nicelib.nicelib: Processing attr '__new__' (<class 'function'>)...
[    INFO]2021-04-30 13:03:08,419 nicelib.nicelib: ...as an ordinary function
[    INFO]2021-04-30 13:03:08,419 nicelib.nicelib: Found NiceObjectDefs: {}
[    INFO]2021-04-30 13:03:08,419 nicelib.nicelib: Found NiceObject subclasses: {}
[    INFO]2021-04-30 13:03:08,420 nicelib.nicelib: Found root sigs: {}
[    INFO]2021-04-30 13:03:08,420 nicelib.nicelib: Found return-handlers: {'return': <RetHandler(name='ret_return')>, 'ignore': <RetHandler(name='ret_ignore')>, '_ret_return': <RetHandler(name='_ret_return')>, '_ret_ignore': <RetHandler(name='_ret_ignore')>}
[    INFO]2021-04-30 13:03:09,022 nicelib: Loading lib tlccs...
[    INFO]2021-04-30 13:03:09,022 nicelib: Loading ._tlccslib from instrumental.drivers.spectrometers...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\s3853339\Miniconda3\envs\test\lib\site-packages\instrumental_lib-0.6-py3.9.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 67, in <module>
    class NiceCCSLib(NiceLib):
  File "C:\Users\s3853339\Miniconda3\envs\test\lib\site-packages\instrumental_lib-0.6-py3.9.egg\instrumental\drivers\spectrometers\thorlabs_ccs.py", line 69, in NiceCCSLib
    _info_ = load_lib('tlccs', __package__)
  File "C:\Users\s3853339\Miniconda3\envs\test\Lib\site-packages\nicelib\__init__.py", line 76, in load_lib
    return LibInfo(lib_module)
  File "C:\Users\s3853339\Miniconda3\envs\test\Lib\site-packages\nicelib\__init__.py", line 18, in __init__
    self._ffi = lib_module.ffi
AttributeError: module 'instrumental.drivers.spectrometers._tlccslib' has no attribute 'ffi'
```

I imagine I'm going in the right direction?

Thanks!

jlozanol, 2021-04-30:
```
(test) C:\Users\s3853339\Miniconda3\envs\test\Lib\site-packages>conda list cffi
# packages in environment at C:\Users\s3853339\Miniconda3\envs\test:
#
# Name                    Version                   Build  Channel
cffi                      1.14.5                   pypi_0    pypi
```

natezb, 2021-04-30:
Hi @jlozanol, and don't worry about being new—in fact you're already formatting your messages very well!

The first error you point out (`ModuleNotFoundError`) is totally normal for this driver. These NiceLib-based drivers initially do not exist, and their modules are generated the first time you try to use them. What's happening here is that module generation is failing.

The "patch" that was mentioned in #120 was actually a change to *NiceLib*, so that's the library you need to install from GitHub, since I haven't made a new release since the bug was fixed last September. You can do the same `pip` install procedure as you did for Instrumental, except you can get the latest NiceLib source this way:

`git clone https://github.com/mabuchilab/NiceLib.git`

jlozanol, 2021-04-30:
OMG! Thx so much @natezb for the clarification!

```
>>> list_instruments()
C:\Users\s3853339\Miniconda3\envs\test\lib\site-packages\instrumental_lib-0.6-py3.9.egg\instrumental\drivers\motion\ecc100.py:24: UserWarning: Driver 'instrumental.drivers.motion.ecc100' is out of date and incompatible with the current Instrumental core. Pull requests fixing this are welcome.
  warnings.warn(
[<ParamSet[NIDAQ] name='myDAQ1' serial=51317026 model=b'NI myDAQ'>, <ParamSet[CCS] usb='USB0::0x1313::0x8087::M00601006::RAW' serial='M00601006' model=<SpecTypes.CCS175: 32903>>]

>>> ccs175.take_data()
[array([0.00129507, 0.00017651, 0.0005573 , ..., 0.00015271, 0.00062869,
       0.00017651]), array([ 498.61413974,  498.77167515,  498.92921807, ..., 1106.38428328,
       1106.55576025, 1106.72723736])]
```

It's finally working!!!

Thanks a lot, this is awesome!!!

closed 2021-04-30 by jlozanol