mabuchilab/Instrumental #139: Modify femto_ferb _check_visa_support method

NitramC, 2021-04-25:
Currently the femto_ferb driver will assign the FemtoFiber class to any instrument that responds to the query '(param-ref system-type)', whether or not that respose is valid. It would be good to modify the _check_visa_support method such that is parses the respose and looks for certain keywords. Unfortunately, the documentation does not provide an example response, so someone who has the insturment would need to make the changes, or one would have to contact Toptica to ask.

Something like this would probably work if we know what the search pattern ???? should be:
```
_SUBCLASS_IDN_SRCPATTERN = {
    # 'subclass': 're.search_pattern'
    'FemtoFiber': '????'
}

def _check_visa_support(visa_rsrc):
    with visa_timeout_context(visa_rsrc, 50):
        try:
            resp = visa_rsrc.query(':version')
            for subclass in _SUBCLASS_IDN_SRCPATTERN.keys():
                if re.search(_SUBCLASS_IDN_SRCPATTERN[subclass], resp) is not None:
                    return subclass
        except:
            pass
    return None
```

natezb, 2021-04-25:
What is the problem you're trying to solve? Do you have some other instrument that's accidentally being recognized as a `FemtoFiber`?

NitramC, 2021-04-25:
Yes, exactly; when running ```list_instruments``` it occured once.

I am working with another serial instrumemt that does not recognize *IDN?. It also will reply to any unrecognized command with somthing like "unknown command xxxx".

natezb, 2021-04-30:
I agree that a change like the one you suggest would be a good one, we just need to find what the string(s) should be.

In the meantime, if you've written a driver for the conflicting instrument, you could give it an earlier `_INST_PRIORITY_` so it is tried before the femto ferb driver.