mabuchilab/Instrumental #140: Communication with Thorlabs uc480 camera

dilrukshi924, 2021-05-08:
I was able to get the current image from a Thorlabs uc480 camera using `instrumental`.  My issue is when I try to adjust the parameters for `grab_image`. I can change `cx` and `left` to any value and get an image. But `cy` and `top` only works if `cy=600` and `top=300`. The purpose is to create a GUI so that the user can select values for these parameters to zoom in/out an image. 

Here is my code
```
import instrumental
from instrumental.drivers.cameras import uc480
from matplotlib.figure import Figure
import matplotlib.pyplot as plt

paramsets = instrumental.list_instruments()
cammer = instrumental.instrument(paramsets[0])

fig1 = Figure(figsize=(6,6))
ax1 = fig1.add_subplot(111)
plt.figure()
framer= cammer.grab_image(timeout='1s',copy=True,n_frames=1,exposure_time='5ms',cx=640,
                                      left=10,cy=600,top=300)
plt.pcolormesh(framer)
```
The above code does not give an image if I choose `cy=600` and `top=10`. Are there any particular value set to be used for these parameters?

natezb, 2021-05-08:
Hi, thanks for the report. What model of camera are you using? Sometimes cameras don't have support for hardware AOI, although if I remember correctly we should have support for software AOI.

Also, some cameras only support some AOI values, e.g. the edges must be powers of 2. It seems a bit strange that x works while y doesn't, but it *could* be the hardware.

dilrukshi924, 2021-05-08:
Hello, this is the camera I'm using 
[https://www.thorlabs.com/thorProduct.cfm?partNumber=DCC1545M](url)

natezb, 2021-05-08:
You might play around with trying different values of `cy` and `top` to see if anything else works, as well as using `top` and `bot` instead.

Also try using `_set_AOI()` and `_get_AOI()` directly to see when and if the AOI can be changed.

natezb, 2021-05-08:
Oh, I just noticed: the camera you're using is only 1024 pixels tall, so it's not big enough to have a rectangle with a center at 600 and top at 10 (the bottom would have to be at 1190). So the `_set_AOI()` silently fails for the y axis.

If this is the issue, we should probably add a warning if the specified AOI is too big.

dilrukshi924, 2021-05-11:
hello, sorry for the delay. I tried different values for the y axis (even cy=500 and top=100 doesn't work. but cy=700 and 400 works.)
How can I find which values to put as cy and top?