mabuchilab/Instrumental #141: UserWarning: Driver 'instrumental.drivers.motion.ecc100' is out of date and incompatible with the current Instrumental core. Pull requests fixing this are welcome.   warnings.warn(

valentinamzztt, 2021-06-08:
I'm very new to Python and I'm working with a DCC3240M from Thorlabs Thorlabs - DCC3240M High-Sensitivity USB 3.0 CMOS Camera, 1280 x 1024, Global Shutter, Monochrome Sensor. I would like to retrieve the images, control the camera settings and so on using Python and I have found the package instrumental which should help me. However, when I run my Python script, I encounter a problem that I was not able to fix. 
This is my code: 
`from thorcam.camera import ThorCam
from instrumental import list_instruments, instrument
import instrumental
from matplotlib.figure import Figure
import matplotlib.pyplot as plt
from instrumental.drivers.cameras import uc480

paramsets = list_instruments() ## camera found
print(paramsets)

paramsets = instrumental.list_instruments()
cammer = instrumental.instrument(paramsets[0])

fig1 = Figure(figsize=(6,6))
ax1 = fig1.add_subplot(111)
plt.figure()
framer= cammer.grab_image(timeout='1s',copy=True,n_frames=1,exposure_time='5ms',cx=640,
                                      left=10,cy=600,top=300)
plt.pcolormesh(framer)

# init camera
instruments = uc480.list_instruments()
cam = uc480.UC480_Camera(instruments[0])
# params
ls_frames = []
print(cam.DEFAULT_KWDS)
cam.start_live_video(framerate = "30Hz")
print (cam.is_open, cam.framerate)
#cam._set_exposure("5ms")
#cam._set_AOI(0, 0, 180, 120)

while (i < total_frames):
    flag = cam.wait_for_frame(timeout = "0 ms")
    if flag:
        i = i + 1
        frame = cam.latest_frame(copy = False)
        frame = frame.astype('uint8')
        ls_frames.append(frame)

class MyThorCam(ThorCam):
    def received_camera_response(self, msg, value):
        super(MyThorCam, self).received_camera_response(msg, value)
        if msg == 'image':
            return
        print('Received "{}" with value "{}"'.format(msg, value))
    def got_image(self, image, count, queued_count, t):
        print('Received image "{}" with time "{}" and counts "{}", "{}"'
              .format(image, t, count, queued_count))

cam = MyThorCam()
cam.start_cam_process()
cam.refresh_cameras()`
The error message I get is the following: 

>  UserWarning: Driver 'instrumental.drivers.motion.ecc100' is out of date and incompatible with the current Instrumental core. Pull requests fixing this are welcome.
  warnings.warn
However, it is able to retrieve the right information regarding the serial number of my camera. Could someone help me to understand where the problem could lie? Thank you

natezb, 2021-06-09:
That's merely a warning, not an error message so you can safely ignore it, plus it's from a driver unrelated to uc480. It will go away if you open the camera directly without `list_instruments()`. For example , once you've opened it once you could save it under an alias using `cam.save_instrument('my-camera')` and load it thereafter using `instrument('my-camera')`.

However, it also looks like you're using some other camera library at the same time? If you're trying to control the same camera by mixing both libraries, that probably won't work.

AsbroBoy, 2022-11-09:
Hey @valentinamzztt, have you figured it out ? I have the same warning when trying to identify my Thorlabs ccs100 spectrometer using `list_instruments()`
Thank you