mabuchilab/Instrumental #142: Is there a possibility of motion tracking ?

shanmukhams, 2021-08-30:
Hi,

Camera - DCX 

I am trying to get a differential video feed (each frame is subtracted from his previous frame) for the detection of the dust particles on the thin films. 

Is there any inbuilt function or any ideas?

I am currently developing using the example /Instrumental/blob/master/examples/camera_gui.py

Thanks

natezb, 2021-08-31:
This isn't the sort of thing that Instrumental would ever do directly. However, the camera methods (e.g. `cam.grab_image()` return numpy arrays, so you can do whatever you want with those. Take a look at `instrumental/gui.py`. You could make a modified version of `CameraView` that holds onto the previous image and subtracts that from the new image before converting it to a pixmap for display.

shanmukhams, 2021-09-01:
thanks for the response and for the idea.

Just to be sure, do I need to change the function _set_pixmap_from_array in instrumental/gui.py?

natezb, 2021-09-01:
There are a number of ways you could do this, but adding something to the top of `_set_pixmap_from_array` would be a simple solution. For example, add this to `__init__`:

```
self._prev_arr = None
```

Then add this at the top of `set_pixmap_from_array`:

```
if self._prev_arr is None
    self._prev_arr = arr
    return
arr, self._prev_arr = (self._prev_arr - arr), arr
```

natezb, 2021-09-01:
One thing to note is that just taking the difference can produce an array with negative values, which can't be directly converted to a pixmap. You'll have to decide how you want to do that mapping. For instance you could produce a color pixmap with one color being positive, another negative. This will require you to change the existing code in `set_pixmap_from_array`.