mabuchilab/Instrumental #143: [cameras.uc480] Unable to change to color mode of my camera

ClassicFan1013, 2021-08-31:
Hi, previously I was using this package for a 8-bit thorlab camera and it worked perfectly. Thanks.
But recently I was trying to another 12-bit cam. But it seems that the images acquired is still 8-bit.
I noticed there's a property named color_mode but it seems that I cannot modify it. Is there any quick solutions for acquiring 12-bit images?

Best regards

natezb, 2021-08-31:
I just looked and the Instrumental code does not yet support 12-bit grayscale, although I think it would be pretty quick to add. I will try to take a look at adding this tonight.

ClassicFan1013, 2021-09-01:
Thank you so much!

natezb, 2021-09-01:
I just pushed some changes to a new branch called `uc480-12bit`. I don't have the hardware to test these changes, so you'll have to try it out and let me know how it goes.

You'll want to uninstall your current Instrumental package, then clone the repo from github, make sure you've switched to the `uc480-12bit` branch, and install from there.