mabuchilab/Instrumental #148: Invalid syntax after import

mdopy, 2021-11-04:
Hello everybody,

i just tried to import with the following statement:
from instrumental import instrument, list_instruments
but always get an invalid syntax error. I have tested the error with python 3.8 and 3.9 in an anaconda environment or with Winpy. 

This is what the Console in Spyder looks like:

> from instrumental import instrument, list_instruments
> Traceback (most recent call last):
> 
>   File "C:\temp\.conda\envs\inst_test\lib\site-packages\IPython\core\interactiveshell.py", line 3444, in run_code
>     exec(code_obj, self.user_global_ns, self.user_ns)
> 
>   File "C:\Users\dolm\AppData\Local\Temp/ipykernel_11456/2322190016.py", line 1, in <module>
>     from instrumental import instrument, list_instruments
> 
>   File "<frozen importlib._bootstrap>", line 1055, in _handle_fromlist
> 
>   File "C:\temp\.conda\envs\inst_test\lib\site-packages\instrumental\__init__.py", line 50, in __getattr__
>     module = __import__(object_origins[name], None, None, [name])
> 
>   File "C:\temp\.conda\envs\inst_test\lib\site-packages\instrumental\drivers\__init__.py", line 24, in <module>
>     from .. import conf
> 
>   File "<frozen importlib._bootstrap>", line 1055, in _handle_fromlist
> 
>   File "C:\temp\.conda\envs\inst_test\lib\site-packages\instrumental\__init__.py", line 55, in __getattr__
>     __import__('instrumental.' + name)
> 
>   File "C:\temp\.conda\envs\inst_test\lib\site-packages\instrumental\conf.py", line 91, in <module>
>     load_config_file()
> 
>   File "C:\temp\.conda\envs\inst_test\lib\site-packages\instrumental\conf.py", line 69, in load_config_file
>     d = literal_eval(value)
> 
>   File "C:\temp\.conda\envs\inst_test\lib\ast.py", line 62, in literal_eval
>     node_or_string = parse(node_or_string, mode='eval')
> 
>   File "C:\temp\.conda\envs\inst_test\lib\ast.py", line 50, in parse
>     return compile(source, filename, mode, flags,
> 
>   File "<unknown>", line 1
>     {'usb': 'USB0::0x1313::0x8081::M00583800::RAW', 'serial': 'M00583800', 'model': <SpecTypes.CCS100: 32897>, 'module': 'spectrometers.thorlabs_ccs', 'classname': 'CCS'}
>                                                                                     ^
> SyntaxError: invalid syntax

natezb, 2021-11-04:
It appears this is a combined bug of the `thorlabs_ccs` driver and the `save_instrument()` method. The syntax error is actually inside your _config_ file, not Instrumental itself. Due to the bug(s), your CCS instrument got saved with invalid syntax. As a short-term workaround, you can probably remove the `'model': <SpecTypes.CCS100: 32897>, ` entry from that line in your config file and your instrument will hopefully still load fine.

I'll have to consider how to deal with this.
1. Loading the config file shouldn't be able to fail so completely. It looks like that call to `literal_eval()` is already wrapped in a try/except, but only catches a `ValueError`. We should either add `SyntaxError` to that, or just catch all exceptions. That way a bad entry will simply be ignored.
2. The issue with the CCS driver is that the `model` parameter is an enum, which isn't being serialized correctly within the config file. Do we instead use a `str` like most other drivers, or figure out the best way to serialize/deserialize the enum. This one I'll have to think about.

mdopy, 2021-11-05:
Thanks for the quick response, it took me a while to find the location of the config file, as it was stored in 
AppData\Local\mabuchilab\instrumental which explains that the error existed independent on the environment I used. Where is it possible to change to location of the config file? 
Furthermore I got the following warning:

> 
> C:\_Work\WPy64-38100\python-3.8.10.amd64\lib\site-packages\visa.py:13: FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.
> 
> The reason for the deprecation is the possible conflict with the visa package provided by the https://github.com/visa-sdk/visa-python which can result in hard to debug situations.
>   warnings.warn(
> C:\_Work\WPy64-38100\python-3.8.10.amd64\lib\site-packages\instrumental\drivers\spectrometers\thorlabs_ccs.py:331: Warning: Data was not successfully acquired: Retrying now
>   warn(Warning("Data was not successfully acquired: Retrying now"))

which I solved by replacing 'from visa import ResourceManager' by 'from pyvisa import ResourceManager' in the thorlabs_ccs.py file

natezb, 2021-11-09:
We bundle a version of the appdirs package to choose the config directory, `appdirs.user_config_dir()` in particular for this file. Appdirs does support setting `XDG_CONFIG_DIRS` to set this manually, but that doesn't work on Windows or Mac systems.

We could possibly support our own custom `$INSTRUMENTAL_CONFIG_DIRS` for this case, which would take precedence over everything else. Does this sound like a reasonable solution?

mdopy, 2021-11-10:
In my opinion it would sufficient to have a note in the documentation, where the config is stored, so that even beginners as I am one find the config quickly

closed 2021-11-10 by mdopy