mabuchilab/Instrumental #158: Another K10CR1 fault?

OpticalDweeb, 2023-04-30:
with a fresh install of instrumental-lib and nicelib  it will not open (or see) the  K10CR1.  Please note this is my first use of instrument-lib and I was hoping for the 'it just works' feature.  I am a moderate nubie to python, but have coded similar libraries in many other languages and systems.

The line from the full log below [    INFO]2023-04-30 14:49:20,559 instrumental.drivers: Error when importing driver module motion._kinesis.isc: <<Cannot find library '('Thorlabs.MotionControl.IntegratedStepperMotors.dll',)'>>

This led me to download the source and look at instrumental/drivers/motion/_kinesis/_build_kinesis.py
Near the top of the libs declaration are:
    "Thorlabs.MotionControl.IntegratedStepperMotors",  

The file  is, as expected,  located in C:\Program Files (x86)\Thorlabs\Kinesis\Thorlabs.MotionControl.IntegratedStepperMotors.dll along with .h and .lib 

I naively added C:\Program Files (x86)\Thorlabs\Kinesis in the PYTHONPATH manager in Spyder which seems unnecessary given the correctly hard-coded file locations.

I get lost at this point, for example trying to find where the instrumental-lib is stored so I can install it...

the log and conf file are below.  Thanks for any help you can offer.  I would be happy with a pointer to RTFM.

Jon

Python 3.9.13 (main, Aug 25 2022, 23:51:50) [MSC v.1916 64 bit (AMD64)]
Type "copyright", "credits" or "license" for more information.

IPython 7.31.1 -- An enhanced Interactive Python.

from instrumental import instrument, list_instruments

from instrumental.log import log_to_screen

log_to_screen()
Out[3]: <RootLogger root (INFO)>

p = list_instruments()
[    INFO]2023-04-30 14:49:20,071 instrumental.drivers: Importing driver module 'cameras.picam'
[    INFO]2023-04-30 14:49:20,074 nicelib.nicelib: Creating class NiceObject...
[    INFO]2023-04-30 14:49:20,075 nicelib.nicelib: Creating class NiceLib...
[    INFO]2023-04-30 14:49:20,076 nicelib.nicelib: Processing attr '__module__' (<class 'str'>)...
[    INFO]2023-04-30 14:49:20,076 nicelib.nicelib: ...as an ordinary class attribute
[    INFO]2023-04-30 14:49:20,076 nicelib.nicelib: Processing attr '__qualname__' (<class 'str'>)...
[    INFO]2023-04-30 14:49:20,077 nicelib.nicelib: ...as an ordinary class attribute
[    INFO]2023-04-30 14:49:20,077 nicelib.nicelib: Processing attr '__doc__' (<class 'str'>)...
[    INFO]2023-04-30 14:49:20,077 nicelib.nicelib: ...as an ordinary class attribute
[    INFO]2023-04-30 14:49:20,078 nicelib.nicelib: Processing attr '_ffi' (<class 'NoneType'>)...
[    INFO]2023-04-30 14:49:20,078 nicelib.nicelib: ...as a special NiceLib attribute
[    INFO]2023-04-30 14:49:20,078 nicelib.nicelib: Processing attr '_ffilib' (<class 'NoneType'>)...
[    INFO]2023-04-30 14:49:20,079 nicelib.nicelib: ...as a special NiceLib attribute
[    INFO]2023-04-30 14:49:20,079 nicelib.nicelib: Processing attr '_defs' (<class 'dict'>)...
[    INFO]2023-04-30 14:49:20,079 nicelib.nicelib: ...as a special NiceLib attribute
[    INFO]2023-04-30 14:49:20,079 nicelib.nicelib: Processing attr '_ret_return' (<class 'nicelib.nicelib.RetHandler'>)...
[    INFO]2023-04-30 14:49:20,080 nicelib.nicelib: ...as a RetHandler
[    INFO]2023-04-30 14:49:20,080 nicelib.nicelib: Processing attr '_ret_ignore' (<class 'nicelib.nicelib.RetHandler'>)...
[    INFO]2023-04-30 14:49:20,080 nicelib.nicelib: ...as a RetHandler
[    INFO]2023-04-30 14:49:20,080 nicelib.nicelib: Processing attr '__new__' (<class 'function'>)...
[    INFO]2023-04-30 14:49:20,081 nicelib.nicelib: ...as an ordinary function
[    INFO]2023-04-30 14:49:20,081 nicelib.nicelib: Found NiceObjectDefs: {}
[    INFO]2023-04-30 14:49:20,081 nicelib.nicelib: Found NiceObject subclasses: {}
[    INFO]2023-04-30 14:49:20,081 nicelib.nicelib: Found root sigs: {}
[    INFO]2023-04-30 14:49:20,081 nicelib.nicelib: Found return-handlers: {'return': <RetHandler(name='ret_return')>, 'ignore': <RetHandler(name='ret_ignore')>, '_ret_return': <RetHandler(name='_ret_return')>, '_ret_ignore': <RetHandler(name='_ret_ignore')>}
[    INFO]2023-04-30 14:49:20,459 nicelib: Loading lib picam...
[    INFO]2023-04-30 14:49:20,460 nicelib: Loading ._picamlib from instrumental.drivers.cameras...
[    INFO]2023-04-30 14:49:20,460 nicelib: Loading build module ._build_picam from instrumental.drivers.cameras...
[    INFO]2023-04-30 14:49:20,461 nicelib.build: Module _picamlib does not yet exist, building it now. This may take a minute...
[    INFO]2023-04-30 14:49:20,462 instrumental.drivers: Error when importing driver module cameras.picam: <<Cannot find library '('Picam.dll',)'>>
[    INFO]2023-04-30 14:49:20,463 instrumental.drivers: Importing driver module 'cameras.pixelfly'
[    INFO]2023-04-30 14:49:20,484 nicelib: Loading lib pixelfly...
[    INFO]2023-04-30 14:49:20,485 nicelib: Loading ._pixelflylib from instrumental.drivers.cameras...
[    INFO]2023-04-30 14:49:20,485 nicelib: Loading build module ._build_pixelfly from instrumental.drivers.cameras...
[    INFO]2023-04-30 14:49:20,486 nicelib.build: Module _pixelflylib does not yet exist, building it now. This may take a minute...
[    INFO]2023-04-30 14:49:20,487 instrumental.drivers: Error when importing driver module cameras.pixelfly: <<Cannot find library '('pf_cam',)'>>
[    INFO]2023-04-30 14:49:20,488 instrumental.drivers: Importing driver module 'cameras.tsi'
[    INFO]2023-04-30 14:49:20,521 instrumental.drivers: Error when importing driver module cameras.tsi: <<ctypes.util.find_library() did not manage to locate a library called 'tsi_sdk'>>
[    INFO]2023-04-30 14:49:20,521 instrumental.drivers: Importing driver module 'cameras.uc480'
[... 132 lines cut ...]
\# This section lists available Instrumental servers and their addresses in
\# host:port form. The keys are server aliases.

#myServer = 0.0.0.0:28265

[instruments] #----------------------------------------------------------------
\# This section includes instruments and their parameters. The keys are aliases,
\# and the values are written as python dicts of instrument parameters. See the
\# examples below for more info.

\# VISA instruments
#myScope = {'visa_address': 'TCPIP::0.0.0.1::INSTR'}
#myAFG = {'visa_address': 'TCPIP::0.0.0.2::INSTR'}

\# NI-DAQ device
#myDAQ = {'nidaq_devname': 'Dev1'}

\# Thorlabs/uEye camera
#myCam = {'ueye_cam_id': 2}

[prefs] #----------------------------------------------------------------------
\# This section is for miscellaneous user-specific preferences.

\# This is a path to the root directory where data files will be saved
data_directory = ~/Data

robotanik, 2023-11-10:
Did you find a solution? I am currently facing the same problem.

robotanik, 2023-11-10:
I thing I have found the problem and was able to fix it by adding the correct path #167 to the system path. Now `ctypes` can find the dll files.