mabuchilab/Instrumental #164: Ecc100 driver

michaelcroquette, 2023-07-04:
I'm trying to update the ecc100.py file and I'm currently able to connect to my instrument but I'm struggling to understand how you define the "Actors" object. Do you have an example of how it used to work (even if outdated) ?

I'm also curious if someone know if the driver will work for positionner without the position reading available.

michaelcroquette, 2023-07-07:
So I manage to make the code work and I have a PR but I can not submit it because I am not a collaborator.

natezb, 2023-07-14:
Are you sure you're unable to create a PR? In case you're not familiar, you should fork the project on GitHub, push your changes to your repo, then issue a PR to our repo. That is generally how we handle external commits, and doesn't require collaborator status as far as I know.